jrowny/cc_sfs #1: Filament Stopped Stuck on Yes

beiyr, 2025-07-29:
As title states, the SFS detects the filament being stopped, even though the response await time is 10 seconds, so it automatically pauses.

jrowny, 2025-08-05:
closing as there was no response and others have successfully got it running

closed 2025-08-05 by jrowny

beiyr, 2025-08-06:
Sorry I haven't been able to respond.
8/5/2025, 7:39:45 PM: Filament movement started
8/5/2025, 7:39:47 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:39:49 PM: Filament movement started
8/5/2025, 7:39:51 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:39:53 PM: Filament movement started
8/5/2025, 7:39:55 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:39:57 PM: Filament movement started
8/5/2025, 7:39:59 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:00 PM: Sending Ping
8/5/2025, 7:40:00 PM: Checking WiFi connection
8/5/2025, 7:40:02 PM: Filament movement started
8/5/2025, 7:40:05 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:06 PM: Filament movement started
8/5/2025, 7:40:08 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:09 PM: Filament movement started
8/5/2025, 7:40:11 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:14 PM: Filament movement started
8/5/2025, 7:40:15 PM: Received status update:
8/5/2025, 7:40:20 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:21 PM: Filament movement started
8/5/2025, 7:40:23 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:24 PM: Filament movement started
8/5/2025, 7:40:26 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:29 PM: Filament movement started
8/5/2025, 7:40:30 PM: Sending Ping
8/5/2025, 7:40:30 PM: Checking WiFi connection
8/5/2025, 7:40:34 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:35 PM: Filament movement started
8/5/2025, 7:40:37 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:40 PM: Filament movement started
8/5/2025, 7:40:42 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:43 PM: Filament movement started
8/5/2025, 7:40:45 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:45 PM: Filament movement started
8/5/2025, 7:40:47 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:57 PM: Filament movement started
8/5/2025, 7:40:59 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:41:00 PM: Sending Ping
8/5/2025, 7:41:00 PM: Checking WiFi connection
8/5/2025, 7:41:02 PM: Received status update:
8/5/2025, 7:41:03 PM: Filament movement started
8/5/2025, 7:41:05 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:41:09 PM: Filament movement started
8/5/2025, 7:41:11 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:41:15 PM: Filament movement started
8/5/2025, 7:41:17 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:41:21 PM: Filament movement started
8/5/2025, 7:41:23 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:41:30 PM: Sending Ping
8/5/2025, 7:41:30 PM: Checking WiFi connection
8/5/2025, 7:41:39 PM: Received status update: