jrowny/cc_sfs #2: Building for esp32-c3

dlarue, 2025-08-01:
I have a number of C3 boards and with the low pin count and performance needs I was wondering if you have insights on why this would not work on the ESP32-C3?

I'm OK with the simple answer. It's a great project and much needed as I've already had a filament knot issue.  I have the  connectors, BTT sensor and esp32-c3 plus crimping tool. WebUI wouldn't recognize the c3. But I have the Arduino v2 setup for esp32 and a CLI platform I configuration.

Thanks.

dlarue, 2025-08-02:
Looks like it'll build but is just a 70 bytes too big to fit in the program space of the esp32-c3.
 Builds to 1,310,798 bytes but only allowed 1,310,720 bytes). I added -0s before ${common.build-flags} but saw no difference in compiled size.

`Compiling .pio/build/lolin_c3_mini/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/lolin_c3_mini/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/lolin_c3_mini/libFrameworkArduino.a
Indexing .pio/build/lolin_c3_mini/libFrameworkArduino.a
Linking .pio/build/lolin_c3_mini/firmware.elf
Retrieving maximum program size .pio/build/lolin_c3_mini/firmware.elf
Checking size .pio/build/lolin_c3_mini/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (1310798 bytes) is greater than maximum allowed (1310720 bytes)
RAM:   [=         ]  12.5% (used 41084 bytes from 327680 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  100.0% (used 1310798 bytes from 1310720 bytes)
`

dlarue, 2025-08-02:
Using the CLI for platformIO for both the esp32-c3 and the esp32-dev board I can compile and upload but the devices go into infinite reboots. Clearly something is different between the uploading the webUI does and what platformIO does.

~/.platformio/penv/bin/platformio run -e esp32-dev --target upload --upload-port /dev/ttyUSB0

booting looks like: 
`rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
E (556) esp_image: Image length 1312592 doesn't fit in partition length 1310720
E (556) boot: OTA app partition slot 0 is not bootable
E (556) esp_image: image at 0x150000 has invalid magic byte (nothing flashed here?)
E (564) boot: OTA app partition slot 1 is not bootable
E (568) boot: No bootable app partitions in the partition table
ets Jul 29 2019 12:21:46
`

The build env for esp32-dev is not unlike esp32-s3:
`[env:esp32-dev]
board = esp32dev
platform = ${common.platform}
framework = ${common.framework}
lib_compat_mode = strict
board_build.filesystem = littlefs
build_flags =
    ${common.build_flags}
lib_deps = 
                ${common.lib_deps}

[env:esp32-build]
board = esp32dev
platform = ${common.platform}
framework = ${common.framework}
lib_compat_mode = strict
board_build.filesystem = littlefs
build_flags =
    ${common.build_flags}
lib_deps = 
                ${common.lib_deps}
extra_scripts = merge_bin.py

`

dlarue, 2025-08-02:
Unfortunately trying to use the webUI locally failed also.

`~/cc_sfs/web$ npm i && npm run dev

up to date, audited 243 packages in 1s

32 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

> web@0.0.0 dev
> node devServer.js

✘ [ERROR] Unexpected end of file in JSON

    ../package.json:1:0:
      1 │ 
        ╵ ^

failed to load config from ~/cc_sfs/web/vite.config.ts
~/cc_sfs/web/node_modules/esbuild/lib/main.js:1477
  let error = new Error(text);
              ^

Error: Build failed with 1 error:
../package.json:1:0: ERROR: Unexpected end of file in JSON
    at failureErrorWithLog (~/cc_sfs/web/node_modules/esbuild/lib/main.js:1477:15)
    at ~/cc_sfs/web/node_modules/esbuild/lib/main.js:946:25
    at runOnEndCallbacks (~/cc_sfs/web/node_modules/esbuild/lib/main.js:1317:45)
    at buildResponseToResult (~/cc_sfs/web/node_modules/esbuild/lib/main.js:944:7)
    at ~/cc_sfs/web/node_modules/esbuild/lib/main.js:971:16
    at responseCallbacks.<computed> (~/cc_sfs/web/node_modules/esbuild/lib/main.js:623:9)
    at handleIncomingPacket (~/cc_sfs/web/node_modules/esbuild/lib/main.js:678:12)
    at Socket.readFromStdout `/cc_sfs/web/node_modules/esbuild/lib/main.js:601:7)
    at Socket.emit (node:events:517:28)
    at addChunk (node:internal/streams/readable:368:12) {
  errors: [Getter/Setter],
  warnings: [Getter/Setter]
}
`

dlarue, 2025-08-03:
I know this is now about building for esp32-dev option but it started as esp32-c3.
I got firmware.bin built with platformIO CLI to load using the webUI update( firmware only ) option.
I had to add the following built ops which I think shrunk it from 99% to 97% and that was enough for the webUI upload to show success.

`build_flags = 
        ${common.build_flags}
        -D CORE_DEBUG_LEVEL=0
        ${env.build_flags}
        -D BOARD_FLASH_SIZE="4M"
        -D LTO
        -D Os
`

jrowny, 2025-08-05:
I don't have any C3 boards, but maybe I'll buy some and then I can add official support. If you'd like to submit a pull request I'll gladly accept it.

dlarue, 2025-08-06:
It builds but like with the WROOM32 I don't have the upload details figured out so the only way I could get running was with the WebUI which works with the WROOM32 and so does the WebUI upload/ElegantOTA if compiled to a small enough image.
Hmm, I don't think I tried either upload on the C3 with the debug level set to 0 so the image was smaller.
I've got like 5 C3 boards and only had 1 of the WROOM32 which I had planned to use in the car OBDII project. I'll order 3-5 more and keep them around because for these low pin count projects they are nice and small.

If I get one of the C3 boards working I'll submit a pull request.

jrowny, 2025-08-06:
you may need to adjust the partition sizes a bit. They're pre-set in platformio but you can override them.