jrowny/cc_sfs #7: Filament Moving Stuck on Yes

beiyr, 2025-08-06:
I apologize for my slow response, everytime I thought about checking it I was already printing. Here are the logs. Note that all the starts are from me pushing the filament into the BTT. I am constantly moving it while it is inside.
`8/5/2025, 7:39:45 PM: Filament movement started
8/5/2025, 7:39:47 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:39:49 PM: Filament movement started
8/5/2025, 7:39:51 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:39:53 PM: Filament movement started
8/5/2025, 7:39:55 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:39:57 PM: Filament movement started
8/5/2025, 7:39:59 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:00 PM: Sending Ping
8/5/2025, 7:40:00 PM: Checking WiFi connection
8/5/2025, 7:40:02 PM: Filament movement started
8/5/2025, 7:40:05 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:06 PM: Filament movement started
8/5/2025, 7:40:08 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:09 PM: Filament movement started
8/5/2025, 7:40:11 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:14 PM: Filament movement started
8/5/2025, 7:40:15 PM: Received status update:
8/5/2025, 7:40:20 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:21 PM: Filament movement started
8/5/2025, 7:40:23 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:24 PM: Filament movement started
8/5/2025, 7:40:26 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:29 PM: Filament movement started
8/5/2025, 7:40:30 PM: Sending Ping
8/5/2025, 7:40:30 PM: Checking WiFi connection
8/5/2025, 7:40:34 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:35 PM: Filament movement started
8/5/2025, 7:40:37 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:40 PM: Filament movement started
8/5/2025, 7:40:42 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:43 PM: Filament movement started
8/5/2025, 7:40:45 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:45 PM: Filament movement started
8/5/2025, 7:40:47 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:40:57 PM: Filament movement started
8/5/2025, 7:40:59 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:41:00 PM: Sending Ping
8/5/2025, 7:41:00 PM: Checking WiFi connection
8/5/2025, 7:41:02 PM: Received status update:
8/5/2025, 7:41:03 PM: Filament movement started
8/5/2025, 7:41:05 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:41:09 PM: Filament movement started
8/5/2025, 7:41:11 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:41:15 PM: Filament movement started
8/5/2025, 7:41:17 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:41:21 PM: Filament movement started
8/5/2025, 7:41:23 PM: Filament movement stopped, last movement detected 2000ms ago
8/5/2025, 7:41:30 PM: Sending Ping
8/5/2025, 7:41:30 PM: Checking WiFi connection
8/5/2025, 7:41:39 PM: Received status update:`

beiyr, 2025-08-06:
If so, is there a way to change it? I'm using jump cables and I can't really move those around.

beiyr, 2025-08-06:
I did the pinout for the first image which you deleted.

beiyr, 2025-08-06:
My wiring is to 13/14 instead of 13/12, as I went off of your first image.

jrowny, 2025-08-06:
that's the problem then, sorry about that! You can also just rebuild the image with a different pin configured by modifying platformio and building, there's no reason it can't work on 13/14.

I'm closing this issue, but feel free to reopen it if changing the pins doesn't work.

closed 2025-08-06 by jrowny

jrowny, 2025-08-06:
you'll need vscode and platformio extension. there's a tons of tutorials out there. Or you can just re-solder the wires onto the correct pins.

beiyr, 2025-08-07:
Do you think I can just keep 14 wired in, but add a jump cable to 12 for it? Or do I need to desolder 14? It's gonna be a pain to do I just want to see if I can do it the easy way.