jrowny/cc_sfs #21: Issues I've run into

SShadow1974, 2025-09-06:
I have run into a few issues. #1 it seems the WebUI doesn't install the latest version of the firmware. I manually updated it which solves the original problems I had. Not going to go into those, since they're old news now. I had commented on your YouTube video, but perhaps that's not the best place to get your attention though you have replied to other comments. *shrug*
So, on to my current issues. My settings are 30000 wait on start up and my timeout is at 6000. I'm powering the ESP via USB and not using the OEM runout sensor.
#1 no matter what I set the startup timeout to, it will always pause somewhere between layer 1 and 2 on every print I've done.
#2 Sometimes when the printer is told to pause, "pausing..." will appear on the display and not go away. I have to power cycle the whole machine, after which it will recover and start printing again.
#3 It will randomly pause for no reason. The filament is running fine and isn't tangled. Unfortunately I don't have a cap of the error from the logs.

Suggestions
Longer log scroll back or a separate tab that retains errors so you can go back and look the error up.
List the version somewhere of the firmware somewhere so you can reference which version is installed on the ESP32

SShadow1974, 2025-09-06:
Well it just paused on me after I posted this. Here is a log:
`9/6/2025, 2:29:30 PM: Sending Ping
9/6/2025, 2:29:30 PM: Received status update:
9/6/2025, 2:29:33 PM: Received status update:
9/6/2025, 2:29:35 PM: Checking WiFi connection
9/6/2025, 2:29:36 PM: Filament movement stopped, last movement detected 6000ms ago
9/6/2025, 2:29:36 PM: Pause condition: 1
9/6/2025, 2:29:36 PM: Filament runout: 0
9/6/2025, 2:29:36 PM: Filament runout pause enabled: 1
9/6/2025, 2:29:36 PM: Filament stopped: 1
9/6/2025, 2:29:36 PM: Time since print start 1616789
9/6/2025, 2:29:36 PM: Is Machine status printing?: 1
9/6/2025, 2:29:36 PM: Print status: 13
9/6/2025, 2:29:36 PM: Pausing print, detected filament runout or stopped
9/6/2025, 2:29:36 PM: Waiting for acknowledgment for command 129 with request ID d94911415e2848ef96eae30c3482233f
9/6/2025, 2:29:36 PM: Command 129 acknowledged (Ack: 0) for request d94911415e2848ef96eae30c3482233f
9/6/2025, 2:29:36 PM: Received expected acknowledgment for command 129
9/6/2025, 2:29:36 PM: Pause condition: 1
9/6/2025, 2:29:36 PM: Filament runout: 0
9/6/2025, 2:29:36 PM: Filament runout pause enabled: 1
9/6/2025, 2:29:36 PM: Filament stopped: 1
9/6/2025, 2:29:36 PM: Time since print start 1617116
9/6/2025, 2:29:36 PM: Is Machine status printing?: 1
9/6/2025, 2:29:36 PM: Print status: 13
9/6/2025, 2:29:36 PM: Pausing print, detected filament runout or stopped
9/6/2025, 2:29:36 PM: Waiting for acknowledgment for command 129 with request ID 5d61ecdcf9b64c4c9f12cbecf18bf307
9/6/2025, 2:29:36 PM: Received status update:
9/6/2025, 2:29:36 PM: Command 129 acknowledged (Ack: 1) for request 5d61ecdcf9b64c4c9f12cbecf18bf307
9/6/2025, 2:29:36 PM: Received expected acknowledgment for command 129
9/6/2025, 2:29:36 PM: Pause condition: 1
9/6/2025, 2:29:36 PM: Filament runout: 0
9/6/2025, 2:29:36 PM: Filament runout pause enabled: 1
9/6/2025, 2:29:36 PM: Filament stopped: 1
9/6/2025, 2:29:36 PM: Time since print start 1617425
9/6/2025, 2:29:36 PM: Is Machine status printing?: 1
9/6/2025, 2:29:36 PM: Print status: 13
9/6/2025, 2:29:36 PM: Pausing print, detected filament runout or stopped
9/6/2025, 2:29:36 PM: Waiting for acknowledgment for command 129 with request ID 30d005d828514955b384dd928d78191d
9/6/2025, 2:29:36 PM: Received status update:
9/6/2025, 2:29:36 PM: Command 129 acknowledged (Ack: 1) for request 30d005d828514955b384dd928d78191d
9/6/2025, 2:29:36 PM: Received expected acknowledgment for command 129
9/6/2025, 2:29:37 PM: Filament movement started
9/6/2025, 2:29:40 PM: Received status update:
9/6/2025, 2:29:43 PM: Received status update:
9/6/2025, 2:29:43 PM: Filament movement stopped, last movement detected 6000ms ago
9/6/2025, 2:29:44 PM: Received status update:
9/6/2025, 2:30:00 PM: Sending Ping
9/6/2025, 2:30:05 PM: Checking WiFi connection
`

I checked the filament, there is no issue with the filament. Moving freely. This time it did not leave the "pausing..." on the screen, so I can resume it. Maybe a "RESUME PRINT" button could be added to the Web interface? Or is that just a one way communication from the ESP to the webbrowser?

SShadow1974, 2025-09-07:
Caught it when it paused again:
9/7/2025, 10:32:22 AM: Received expected acknowledgment for command 129
9/7/2025, 10:32:22 AM: Pause condition: 1
9/7/2025, 10:32:22 AM: Filament runout: 0
9/7/2025, 10:32:22 AM: Filament runout pause enabled: 1
9/7/2025, 10:32:22 AM: Filament stopped: 1
9/7/2025, 10:32:22 AM: Time since print start 446673
9/7/2025, 10:32:22 AM: Is Machine status printing?: 1
9/7/2025, 10:32:22 AM: Print status: 13
9/7/2025, 10:32:22 AM: Pausing print, detected filament runout or stopped
9/7/2025, 10:32:22 AM: Waiting for acknowledgment for command 129 with request ID 039b73bd2ce340c397bde9d2c46d9a50
9/7/2025, 10:32:23 AM: Command 129 acknowledged (Ack: 1) for request 039b73bd2ce340c397bde9d2c46d9a50
9/7/2025, 10:32:23 AM: Received expected acknowledgment for command 129
9/7/2025, 10:32:23 AM: Pause condition: 1
9/7/2025, 10:32:23 AM: Filament runout: 0
9/7/2025, 10:32:23 AM: Filament runout pause enabled: 1
9/7/2025, 10:32:23 AM: Filament stopped: 1
9/7/2025, 10:32:23 AM: Time since print start 446985
9/7/2025, 10:32:23 AM: Is Machine status printing?: 1
9/7/2025, 10:32:23 AM: Print status: 13
9/7/2025, 10:32:23 AM: Pausing print, detected filament runout or stopped
9/7/2025, 10:32:23 AM: Waiting for acknowledgment for command 129 with request ID 80cf6a37b5d241239a35ab2154764eef
9/7/2025, 10:32:23 AM: Received status update:
9/7/2025, 10:32:23 AM: Command 129 acknowledged (Ack: 1) for request 80cf6a37b5d241239a35ab2154764eef
9/7/2025, 10:32:23 AM: Received expected acknowledgment for command 129
9/7/2025, 10:32:23 AM: Pause condition: 1
9/7/2025, 10:32:23 AM: Filament runout: 0
9/7/2025, 10:32:23 AM: Filament runout pause enabled: 1
9/7/2025, 10:32:23 AM: Filament stopped: 1
9/7/2025, 10:32:23 AM: Time since print start 447289
9/7/2025, 10:32:23 AM: Is Machine status printing?: 1
9/7/2025, 10:32:23 AM: Print status: 13
9/7/2025, 10:32:23 AM: Pausing print, detected filament runout or stopped
9/7/2025, 10:32:23 AM: Waiting for acknowledgment for command 129 with request ID a8a30eedd7324aa08255d4b8ffc76f7a
9/7/2025, 10:32:23 AM: Received status update:
9/7/2025, 10:32:23 AM: Command 129 acknowledged (Ack: 1) for request a8a30eedd7324aa08255d4b8ffc76f7a
9/7/2025, 10:32:23 AM: Received expected acknowledgment for command 129
9/7/2025, 10:32:25 AM: Filament movement started
9/7/2025, 10:32:26 AM: Received status update:
9/7/2025, 10:32:29 AM: Received status update:
9/7/2025, 10:32:30 AM: Sending Ping
9/7/2025, 10:32:30 AM: Checking WiFi connection
9/7/2025, 10:32:31 AM: Filament movement stopped, last movement detected 6000ms ago
9/7/2025, 10:32:32 AM: Received status update:
9/7/2025, 10:32:36 AM: Filament movement started
9/7/2025, 10:32:37 AM: Received status update:
9/7/2025, 10:32:37 AM: Received status update:
9/7/2025, 10:32:37 AM: Received status update:
9/7/2025, 10:32:37 AM: Received status update:
9/7/2025, 10:32:37 AM: Received status update:
9/7/2025, 10:32:45 AM: Received status update:
9/7/2025, 10:32:57 AM: Received status update:
9/7/2025, 10:33:00 AM: Sending Ping
9/7/2025, 10:33:00 AM: Received status update:
9/7/2025, 10:33:00 AM: Checking WiFi connection
9/7/2025, 10:33:02 AM: Filament movement stopped, last movement detected 6000ms ago
Logs are automatically refreshed every 5 seconds.

s3anami, 2025-09-13:
> Generally a good ol fashioned search will be better than Gemini, https://github.com/Chirimorin/CCOrcaLayerNumberFix

So the runout sensor will work if that G-code is added?

SShadow1974, 2025-09-13:
> > Generally a good ol fashioned search will be better than Gemini, https://github.com/Chirimorin/CCOrcaLayerNumberFix
> 
> So the runout sensor will work if that G-code is added?

That has not been my experience so far. That GCode was already present in my settings... And I am still having those issues.

jrowny, 2025-09-13:
I did not modify the code to use the current layer yet, there's a PR where I made the changes but it's not merged as I need to do more testing.

If you're able to build that branch and test it, it would be helpful. 

At some point I'll set up branch builds, but I haven't had time.

jrowny, 2025-09-13:
It's not bad, download vs code, add the platformio extension under the extensions tab, maybe a github extension (or use a github client), download the project, and hit build. If it works, plug in your esp32 and hit build & upload. 

Now using the GitHub client checkout the branch and build&upload again. 

Then congratulations, you're a programmer! 

And you can also just build and upload to the ota screen if you don't want to unplug it from it's home.

jrowny, 2025-09-13:
> Ok, so I've got the Visual Studio Code and the PlatformIO installed. I think I even managed to clone the layerfix repository to my local computer and open it in PlatformIO. Now, I have zero idea where to go from there.

Open the folder in vs code, You should have a platformio tab on the left-hand side of the screen, looks like an alien head. Click it, select the esp32s3-dev (assuming your on an s3), and press build in the list that shows up. Then you can go into the .pio folder to find the build that it made, or you can click upload if you have it plugged in.

SShadow1974, 2025-09-13:
Ok, so the file I want is the firmware.bin file in the PIO folder, correct?

SShadow1974, 2025-09-13:
Is there a way to doublecheck that I have the correct repository with the layerfix changes?

jrowny, 2025-09-13:
> Ok, so the file I want is the firmware.bin file in the PIO folder, correct?
> 
Yes, specifically the one in the folder labeled esp32s3-dev 

And to check you have the umright branch, look in my PR and see the code I changed that uses current layer in the if statement, check that it matches what you have. Feel free to tweak that logic and experiment.

SShadow1974, 2025-09-13:
Holy crap, I'm a coder! LMAO. Uploading the firmware.bin to the ESP32 now.
Now exactly how do I test the changes you made, beyond just doing a print? Should I reset my timeouts from 15000 and 30000? I'd like to be able to give you the best test results I can.

jrowny, 2025-09-13:
I usually print a small gridfinity bin because I know I'll use it and it only takes 15 minutes.

I would see if you can get through a whole print at your current settings and then reduce the value until it starts pausing

SShadow1974, 2025-09-13:
Roger that. I always have something to print, so I'll do some and we'll see what happens. Just doing a self check now because I replaced the extruder cover with the one from SE3D. I'll report back with my findings.

SShadow1974, 2025-09-14:
*crossing fingers*
5.5hrs into current print, no pauses yet. It's a slow print, going at "silent" speed because I was having issues with layer adhesion with this ASA... So we'll see where things end up.

SShadow1974, 2025-09-15:
A completed 20hr print, zero pauses. Now, the question is, will it pause the print if I purposely stop the filament. :) Just a couple more tests and then I'll start reducing the times and see if we can get er back to something less than my current 15000/30000.

SShadow1974, 2025-09-15:
Damn, layer 1 pause on this current print. It does pause when filament stops, so that a plus. But on this print, it did pause on layer 1. I haven't changed my times yet, either. So it has not fixed that problem, unfortunately.

SShadow1974, 2025-09-15:
Woke up this morning and it had paused again. So whatever the issue is, it's not fixed as of yet.

SShadow1974, 2025-09-15:
And it paused on the final layer of the print again, hit resume and the extruded wipes back and forth and brings up the print finished dialog. So there is no change with the layerfix branch.

jrowny, 2025-09-15:
> And it paused on the final layer of the print again, hit resume and the extruded wipes back and forth and brings up the print finished dialog. So there is no change with the layerfix branch.

I have no idea what could be taking so long that it's not registering movement, are you seeing any under extrusion on your prints? Maybe it actually is catching some drag somewhere? You can compile firmware now so you can add more debugging.

We're the two prints done at different speeds?

I'm also working on another strategy to try and communicate flow rate to the software

SShadow1974, 2025-09-15:
And the current print finished without any further pauses. I'm going to take apart my sensor now and see if I see anything.

SShadow1974, 2025-09-16:
Ok, I took it apart. The only thing I saw is the light emitting diode was not straight off the board, it was bent a little upwards. So I bent it straight, tested it with a piece of filament and everything seems to be fine. The filament isn't slipping in the (for the lack of better words) gears and the web interface as well as the light on the SFS show when the filament is moving and when it's not. I even tried moving the filament VERY slowly - like I could feel the clicks when moving the filament and I would move it just one click and it would register the movement. So I'm now pretty sure it's not a hardware issue - unless bending that diode straight was the issue. I'll be doing a print later (waiting for some glue to dry) and will report back - again. :) Wish I could be more help.

SShadow1974, 2025-09-16:
And paused again, this time with a - ms reading. I think I may go back to the beta firmware. It worked better, I think.
9/16/2025, 12:23:51 AM: Filament movement stopped, last movement detected -335ms ago
9/16/2025, 12:23:51 AM: Pause condition: 1
9/16/2025, 12:23:51 AM: Filament runout: 0
9/16/2025, 12:23:51 AM: Filament runout pause enabled: 1
9/16/2025, 12:23:51 AM: Filament stopped: 1
9/16/2025, 12:23:51 AM: Time since print start 760134
9/16/2025, 12:23:51 AM: Is Machine status printing?: 1
9/16/2025, 12:23:51 AM: Print status: 13
9/16/2025, 12:23:51 AM: Pausing print, detected filament runout or stopped
9/16/2025, 12:23:51 AM: Waiting for acknowledgment for command 129 with request ID f952d314b557492a9c89639fbcbbdcea
9/16/2025, 12:23:51 AM: Filament movement started
9/16/2025, 12:23:51 AM: Command 129 acknowledged (Ack: 0) for request f952d314b557492a9c89639fbcbbdcea
9/16/2025, 12:23:51 AM: Received expected acknowledgment for command 129
9/16/2025, 12:23:52 AM: Received status update:
9/16/2025, 12:23:53 AM: Received status update:
9/16/2025, 12:23:56 AM: Received status update:
9/16/2025, 12:23:58 AM: Received status update:
9/16/2025, 12:24:05 AM: Sending Ping
9/16/2025, 12:24:09 AM: Filament movement stopped, last movement detected 15000ms ago

SShadow1974, 2025-09-16:
I don't know if any more of these log reports are of any use to you... But I'm back on the 0.2.2 firmware and it's still pausing. I've gone ahead and ordered another board from AliExpress. Until then, this is pretty much useless to me. :(
9/16/2025, 12:47:50 AM: Filament movement stopped, last movement detected -44ms ago
9/16/2025, 12:47:50 AM: Pause condition: 1
9/16/2025, 12:47:50 AM: Filament runout: 0
9/16/2025, 12:47:50 AM: Filament runout pause enabled: 1
9/16/2025, 12:47:50 AM: Filament stopped: 1
9/16/2025, 12:47:50 AM: Time since print start 353625
9/16/2025, 12:47:50 AM: Is Machine status printing?: 1
9/16/2025, 12:47:50 AM: Print status: 13
9/16/2025, 12:47:50 AM: Pausing print, detected filament runout or stopped
9/16/2025, 12:47:50 AM: Waiting for acknowledgment for command 129 with request ID 5b056ee9ba7f4274a9d397f2620a6be6
9/16/2025, 12:47:50 AM: Filament movement started
9/16/2025, 12:47:50 AM: Sending Ping
9/16/2025, 12:47:50 AM: Command 129 acknowledged (Ack: 0) for request 5b056ee9ba7f4274a9d397f2620a6be6
9/16/2025, 12:47:50 AM: Received expected acknowledgment for command 129
9/16/2025, 12:47:51 AM: Received status update:
9/16/2025, 12:47:52 AM: Received status update:
9/16/2025, 12:47:55 AM: Received status update:
9/16/2025, 12:47:58 AM: Received status update:
9/16/2025, 12:48:00 AM: Received status update:
9/16/2025, 12:48:07 AM: Filament movement stopped, last movement detected 15000ms ago

SShadow1974, 2025-09-25:
Well I just recieved a new SFS2.0. I'm just waiting on another ESP32 S3. I'm going to set up another sensor to rule out hardware issues. It simply does not work the way it is. I'm up to 30000 on both timeout settings and it will still randomly pause. On some prints, it'll print the whole thing. Other prints it stops after the first layer and then finishes. Other times it will pause over and over and over again. I just can't figure a common denominator with this. So I'm just going to rule out the issue being on my end.
Have you heard from anyone else that doesn't have ANY issues? Or is my experience common amoung all?

rhouri, 2025-09-28:
I have that issue as well. Restored the original filament sensor and issue disappeared.

SShadow1974, 2025-10-14:
No further updates on this, eh?